pebble-dev/RebbleOS #19: faster log output

ginge, 2017-05-14:
Log formatting crude. Needs cleanup

jwise, 2017-05-15:
PebbleOS has a pretty good architecture for this in general.  I think they have strings in debug builds that get collapsed to filename/line number in prod builds.  As a bonus, that means that they have a compressed log format that can get splatted out over Bluetooth quickly -- or a log format that can get written to flash without taking up too much space.

I think a good first cut at this would be to see if we can get rid of (!) the printf() function from the codebase entirely.  Then, anyone doing output at least is forced to get marshaled through a common log entry point...

eliedrian, 2019-10-23:
I'm working my way to removing printf's and replacing them with what I think are appropriate <type>_LOG's

jwise, 2019-11-03:
That'd be great!  That would also pipeclean for #80.