pebble-dev/RebbleOS #28: Re-implement splash screen

ginge, 2017-05-29:
The hard coded splash was removed. Re-do the boot splash:
* Create a new resource pack for the system resources, and upload. The splash should be in there in png.
* move display init to after flash init
* add hooks to load the resource from flash using the driver api
* display splash

ramonschepers, 2017-08-11:
Regarding the term "resource pack" what does this need to contain separate from the splash image? A bit more info would be usefull :) (I happen to have made some GUI designs before, so I might help with this)

ramonschepers, 2017-08-11:
right now the rebble.io icon is fairly monotone, which i both like and dislike, i tried to get what i had in mind onto a png without too much blingbling and such. i personally think this looks more modern :)
[image]

jneubrand, 2017-08-11:
[image]
Quick mockup of a platform-agnostic image using the closest matches to the rebble color scheme within the 64-color palette (turning it into b/w should be easy; I'd suggest a black background) — any center crop of this should work. Added benefit: this version could hypothetically animate to the default watchface after startup.

Avamander, 2017-09-02:
What do you think about just subtly changing the current default boot splash's `P` in `Pebble` to `R` as in `Rebble`?

jwise, 2017-09-03:
@avamander I think it would probably be better to more assertively be of our own design.  That is to say, it'd be better if we had something that wasn't "trying to be Pebble" (or, to the chagrin of their lawyers, "easily confusable with Pebble"...).

Avamander, 2017-09-03:
@jwise That's going to be hard if RebbleOS runs on Pebbles, Pebble apps and like PebbleOS. Just saying "RebbleOS" would work too, my idea is that it should be minimalist as possible, no lines, no nothing.

Stubenhocker1399, 2018-01-23:
https://github.com/ginge/FreeRTOS-Pebble/pull/55
Can be closed. 😄

closed 2018-01-28 by jwise