pebble-dev/RebbleOS #29: Neographics licensing, merge

jwise, 2017-05-30:
blocks pebble-dev/RebbleOS#24

jneubrand, 2017-05-30:
I agree to release all code and documentation I wrote for neographics (in effect, everything in pebble-dev/neographics as of commit e305f8c3, except for the `docs` folder, as that contains some content derived from the doxygen project) under the MIT license as found on https://opensource.org/licenses/MIT.

I'll merge in changes from ginge/FreeRTOS-pebble and replace GPL with the MIT license in the repository pending approval from *every person who modified the neographics code found therein*. (@ginge @simon @m0n1s, according to my count.)

To avoid any possible future legal trouble, I'd like to ask the people mentioned above to affirm via a comment stating "I release my modifications to neographics under the MIT license."

Thanks!

m0n1s, 2017-05-31:
I release my modifications to neographics under the MIT license. Wuhuu : D

ginge, 2017-06-04:
I release my modifications to neographics under the MIT license.

jwise, 2017-08-13:
@jneubrand do you know which commit @simon is responsible for?  I cannot seem to find it; if you let me know, I'll send him mail to follow up so we can get this closed

jneubrand, 2017-08-13:
@jwise My bad. I hadn't looked at the [diff](https://github.com/ginge/FreeRTOS-Pebble/commit/862b6d0fbe17cf1268c6c2159b32af03effd50eb) closely enough—he seems to only have moved some code & renamed a function (which, to my understanding, shouldn't fall under US copyright law.) Since the change comes from the `ui` directory, it falls under the MIT License (attributed to The RebbleOS Authors.)

I'll get these changes merged in today.

jneubrand, 2017-08-13:
Ah, looks like there's also [this](https://github.com/ginge/FreeRTOS-Pebble/commit/cb43b6b0b2cd8f76c4fb991572c5f53f3aeb4576) commit, whose changes to the neographics source _could_ be interpreted as being made with the assumption of release under GPL. _However_, the changes made are minuscule.

jneubrand, 2017-08-15:
6ac5c9b2 in neographics implements necessary changes (caveat: great centering function is called `grect_center_point` again). As such, this is pending closure as soon as rebbleos builds with the neographics master.

jneubrand, 2018-02-08:
@Stubenhocker1399 @CTKRocks @Helco @zbsz Please agree to release your changes under the MIT license by copy-pasting the following text & replying to this issue with it.

"I release my modifications to neographics under the MIT license."

Stubenhocker1399, 2018-02-08:
I release my modifications to neographics under the MIT license.

😄

zbsz, 2018-02-08:
I release my modifications to neographics under the MIT license.

CTKRocks, 2018-02-08:
I release my modifications to neographics under the MIT license.

closed 2018-02-11 by ginge