pebble-dev/RebbleOS #33: snowy: power_incr leak in display resulting in overflow after some frames

jwise, 2018-01-07:
```
(gdb) bt
#0  panic (s=0x80351bc "Assertion failed: !((incr == 1) && (statep[i] == 0xFF)) && \"stm32_power_incr overflow\" (hw/drivers/stm32_power/stm32_power.c:109)") at rcore/debug.c:15
#1  0x0802f688 in stm32_power_incr (reg=STM32_POWER_AHB1, domain=64, incr=1) at hw/drivers/stm32_power/stm32_power.c:109
#2  0x0802f76a in stm32_power_request (reg=STM32_POWER_AHB1, domain=64) at hw/drivers/stm32_power/stm32_power.h:52
#3  0x0802fd08 in _snowy_display_cs (enabled=1 '\001') at hw/platform/snowy_family/snowy_display.c:397
#4  0x08030018 in _snowy_display_start_frame (xoffset=0 '\000', yoffset=0 '\000') at hw/platform/snowy_family/snowy_display.c:553
#5  0x0803031a in hw_display_start_frame (xoffset=0 '\000', yoffset=0 '\000') at hw/platform/snowy_family/snowy_display.c:705
#6  0x08017da0 in _display_start_frame (xoffset=0 '\000', yoffset=0 '\000') at rcore/display.c:74
#7  0x08017e7c in _display_thread (pvParameters=0x0) at rcore/display.c:134
#8  0x08008be0 in ?? ()
```

Stubenhocker1399, 2018-01-27:
I'm not 100% about this, since I don't exactly understand what this is for, but it seems to me that line 551 in snowy_display.c is at fault:
https://github.com/ginge/FreeRTOS-Pebble/blob/master/hw/platform/snowy_family/snowy_display.c#L551

request_clocks() and release_clocks() (https://github.com/ginge/FreeRTOS-Pebble/blob/master/hw/platform/snowy_family/snowy_display.c#L227) both have STM32_POWER_AHB1, although with the second argument of RCC_AHB1Periph_GPIOG, where as line 551 has display.clock_display.

Now, either line 551 needs to be commented out or a release for it needs to be added, in my understanding the least. :sweat_smile:

ginge, 2018-01-27:
"Now, either line 551 needs to be commented "
This. Good spot. If you want to test and do a PR, that would be super!
Thanks!

closed 2018-01-28 by ginge