pebble-dev/RebbleOS #55: make fails

Allen-B1, 2018-03-21:
I put the `snowy_fpga.bin` in Resources and then ran `make` and got the following message at the end:
```
[snowy] CC hw/platform/snowy/snowy.c
[snowy] CC hw/platform/snowy/snowy_bluetooth.c
[snowy] FPGA Resources/snowy_fpga.bin
[snowy] LD build/snowy/tintin_fw.elf
6772 bytes of CCRAM available.
22332 bytes of RAM available for heap.
362704 bytes of flash unused.
OBJCOPY build/snowy/tintin_fw.bin
[snowy] PBZ build/snowy/snowy.pbz
usage: mkpbz.py [-h] [-p PLATFORM] [-c COMMIT] [-v VERSION] [-l LICENSE]
                firmware respack pbz
mkpbz.py: error: argument -c/--commit: expected 1 argument(s)
Makefile:156: recipe for target 'build/snowy/snowy.pbz' failed
make: *** [build/snowy/snowy.pbz] Error 2
```

jwise, 2018-03-21:
Looks like something is weird with your git setup. What do you get when you run `git describe --always --dirty --exclude '*'` in the `Pebble-FreeRTOS` directory?

Allen-B1, 2018-03-22:
I get `error: unknown option 'exclude'`. I think my version of `make` is too old

Allen-B1, 2018-03-22:
Update git, now works. Thanks!

Allen-B1, 2018-03-22:
I got another issue:
```
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: build/chalk/tintin_fw.elf section `.ccmram' will not fit in region `RAM2'
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: region `RAM2' overflowed by 1436 bytes
collect2: error: ld returned 1 exit status
Makefile:156: recipe for target 'build/chalk/tintin_fw.elf' failed
make: *** [build/chalk/tintin_fw.elf] Error 1
```

Allen-B1, 2018-03-23:
wait chalk isn't supported, sorry didn't read that

closed 2018-03-23 by Allen-B1