pebble-dev/RebbleOS #62: graphics context smashed

ginge, 2018-05-30:
neographics context is created in app_malloc on thread start. This could cause an issue where the overlay thread is using the draw context while an app is loaded. The app loading deletes the memory that drawing requires.

Allocate the context at the global level (will require changes to ngfx) or statically declare this

see appmanager.c:598

Helco, 2018-05-31:
I would propose introducing multiple context instances, so the app, overlay, OS, etc. all have their own n_GContext based on the same framebuffer. If locking the framebuffer to only one writer is necessary, we could use the already existing parenting feature (meant for PebbleOS), which uses `graphics_capture_frame_buffer` of the parent context. Additionally this approach would help in: 

- A not yet documented issue of overlay setting context attributes (e.g. fill color) and thus changing the drawing of the app (and vice versa).
- A proposal for the future would be to move the clipping out of RebbleOS to ngfx and allow for an easy way to implement `layer_set_clips` and prevent the app messing with the area of the overlay

ginge, 2018-06-15:
Locking shouldn't be a problem for now, but a good idea. Due to the way that the main app and overlay are drawn, it shouldn't be possible for an app to draw over the overlay. We draw in this sequence: App/Overlay draw request => app thread draw => overlay thread draw. it means a full page layer re-draw from app when overlay changes, but that's ok. We need to be mindful of the overlay rect changing, such as reducing in size, thus requiring a full app layer draw. Currently this is a requirement that all layers re-draw. It's just simpler.

I agree with all of your suggestions!
It would help to break the draw chain above to become a little more async. But with the minor challenges of frame by frame rect changes.
Related: A per thread context would help with my preference of the draw x,y offset handling to be done in ngfx. Such as setting the offset in the context before draw commands. I'm sure theres a more elegant way to do that.

Good feedback

ginge, 2019-09-01:
Resolved by 8314dde888cbbf66197c944b49a220d1402a43ad

closed 2019-09-01 by ginge