pebble-dev/RebbleOS #64: Trying to add an app in Pebble Store on Apple Phone

rachelbeam90, 2018-08-10:
When trying to add an app (specifically "SmartWatch by Smart Monitor") to a user's iPhone, after searching for it within the Pebble Watch app store and pressing "add", users are getting the following error message in red:

cannot add app to locker

This issue is not occurring on Android phones.

jwise, 2018-08-10:
this is not a RebbleOS bug -- try opening against pebble-dev/rebble-appstore-api (or asking in Discord #rebble-help)?

closed 2018-08-10 by jwise

rachelbeam90, 2018-08-10:
Hello,

thank you for your prompt response. Can you elaborate on "try opening
against pebble-dev/rebble..."

KonradIT, 2018-08-10:
@rachelbeam90 open an issue in the repository https://github.com/pebble-dev/rebble-appstore-api