pebble-dev/RebbleOS #67: Urgent need...please help: Can I extract a custom watchface directly from a Pebble Time Steel?

madmanmoose, 2018-10-14:
Hi.

My son died 2 years ago; and, my last Christmas gift to him was a Pebble Time Steel, that I still have.
He luv'd the watch, and created a custom watchface that was HIS ID for the watch.

Before Fitbit stopped hosting the Pebble servers, I made a point of wearing his watch at least once every couple of weeks. However, if I move his watch over to Rebble now, I will lose that watchface. 
I've searched through his computer/tablet/smartphone...but, can't find the original file. 

Is there any way at all to extract that watchface from the Pebble Time Steel itself?

jwise, 2018-10-14:
I'm so sorry to hear that.

You might be able to extract it from his phone by doing a backup of the Pebble app on the phone. In the worst case, I have a procedure for destructively pulling the firmware from Time-series devices (i.e., getting the firmware and apps out of the device, but destroying the device in the process). I think we should be able to do better, though - it should be cached in the app's private storage on the phone.

I'm on mobile right now, but please send me a mail - joshua@joshuawise.com - and we'll see what we can come up with.

closed 2018-10-14 by jwise

jwise, 2018-11-05:
Hmm, looks like my previous mail didn't make it to you.  I tried again (if it didn't arrive, maybe try your spam folder?).