pebble-dev/RebbleOS #77: Remove remaining PebbleOS resources

jwise, 2019-06-09:
Currently, RebbleOS sucks in PebbleOS resource packs; on tintin-family devices, it sucks in the following:

- [ ] RESOURCE_ID_AGENCY_FB_60_THIN_NUMBERS_AM_PM
- [ ] RESOURCE_ID_AGENCY_FB_60_NUMBERS_AM_PM
- [ ] RESOURCE_ID_AGENCY_FB_36_NUMBERS_AM_PM
- [ ] RESOURCE_ID_GOTHIC_09
- [ ] RESOURCE_ID_GOTHIC_14
- [ ] RESOURCE_ID_GOTHIC_14_BOLD
- [ ] RESOURCE_ID_GOTHIC_18
- [ ] RESOURCE_ID_GOTHIC_18_BOLD
- [ ] RESOURCE_ID_GOTHIC_24
- [ ] RESOURCE_ID_GOTHIC_24_BOLD
- [ ] RESOURCE_ID_GOTHIC_28
- [ ] RESOURCE_ID_GOTHIC_28_BOLD
- [ ] RESOURCE_ID_GOTHIC_36
- [ ] RESOURCE_ID_GOTHIC_36_BOLD
- [ ] RESOURCE_ID_BITHAM_18_LIGHT_SUBSET
- [ ] RESOURCE_ID_BITHAM_34_LIGHT_SUBSET
- [ ] RESOURCE_ID_BITHAM_30_BLACK
- [ ] RESOURCE_ID_BITHAM_42_BOLD
- [ ] RESOURCE_ID_BITHAM_42_LIGHT
- [ ] RESOURCE_ID_BITHAM_34_MEDIUM_NUMBERS
- [ ] RESOURCE_ID_BITHAM_42_MEDIUM_NUMBERS
- [ ] RESOURCE_ID_ROBOTO_CONDENSED_21
- [ ] RESOURCE_ID_ROBOTO_BOLD_SUBSET_49
- [ ] RESOURCE_ID_DROID_SERIF_28_BOLD
- [ ] RESOURCE_ID_LECO_20_BOLD_NUMBERS
- [ ] RESOURCE_ID_LECO_26_BOLD_NUMBERS_AM_PM
- [ ] RESOURCE_ID_LECO_32_BOLD_NUMBERS
- [ ] RESOURCE_ID_LECO_36_BOLD_NUMBERS
- [ ] RESOURCE_ID_LECO_38_BOLD_NUMBERS
- [ ] RESOURCE_ID_LECO_28_LIGHT_NUMBERS
- [ ] RESOURCE_ID_LECO_42_NUMBERS
- [ ] RESOURCE_ID_EMOJI_14
- [ ] RESOURCE_ID_EMOJI_18
- [ ] RESOURCE_ID_EMOJI_24
- [ ] RESOURCE_ID_EMOJI_28
- [ ] RESOURCE_ID_FONT_FALLBACK

We need replacements for these fonts in order to redistribute RebbleOS on non-Pebble devices.  Renaissance should replace some of these; what about the rest?

jneubrand, 2019-06-09:
Oh hi! I can probably do a handful of these :)

jneubrand, 2019-06-09:
[IANAL]  since these are all bitmap fonts, their actual shape is not covered by US copyright laws as far as I can tell. If we extract all letters as .pngs, or whatever, and recompile a font with the same metrics, we should be clear... _in the US_. 

For other countries this is a bit more complicated but I think I still have the metrics-extracting scripts ready so using the ascii representation & compiling new fonts should be alright. (For roboto & droid specifically, we should use the direct-recompile approach.)

Not sure about agency

For LECO, I’d like to consider doing something a bit different (eg with serifs) but stick to the blocky design overall.

For Bitham, idk. That one will take a good amount of time.

jwise, 2019-06-09:
I was hoping you'd jump on this one! :-)

For some of these it might even be okay to punt -- if we don't need them to get the system up and running, we can defer it for quite some time, probably.

jneubrand, 2019-06-13:
BTW, I designed the font in [this watchface][1] and would be happy to donate it to the cause :) unfortunately I no longer have the raw code, do you know how to fetch app tarballs from the currently live store?

[1]: https://apps.rebble.io/en_US/application/596e5098461a8dd0c0000090?native=true&query=neubrand&section=watchfaces&dev=true

CennoxX, 2019-08-29:
@jneubrand The query parameter is not named dev, it's dev_settings. If you open [this link](
https://apps.rebble.io/en_US/application/596e5098461a8dd0c0000090?native=true&query=neubrand&section=watchfaces&dev_settings=true), you'll get a link to download the .pbw at the bottom.

jwise, 2019-09-14:
Renaissance fonts are now the default!

jwise, 2019-09-14:
Ok, I've submitted a PR to the metrics in Renaissance for metrics for _all_ Pebble fonts.

jneubrand, 2019-09-15:
@jwise thanks a ton!

leo60228, 2021-08-02:
> Gotham rasterizations. Maybe there's some similar SIL-OFL font (Montserrat comes to mind, though its metrics are screwy?) that we can base the new sprites on?

VMware's [Clarity City](https://github.com/vmware/clarity-city) seems like it could be a good choice. It's derived from Metropolis, which is [very similar to Gotham](https://graphicdesign.stackexchange.com/a/146825).