pebble-dev/RebbleOS #79: tintin_fw.elf section `.ccmram' will not fit in region `RAM2' region `RAM2' overflowed by 3412 bytes

Bystroushaak, 2019-07-26:
I've tried to follow the [guide for Debian](https://github.com/pebble-dev/RebbleOS/blob/master/docs/debian_build.md) and I am getting following error:

```make
...
(.env) a@a-VirtualBox:~/RebbleOS$ make
VERSION build/version.c
[snowy] CC build/version.c
[snowy] LD build/snowy/tintin_fw.elf
4796 bytes of CCRAM available.
30244 bytes of RAM available for heap.
595644 bytes of flash unused.
OBJCOPY build/snowy/tintin_fw.bin
[snowy] PBZ build/snowy/snowy.pbz
[tintin] CC build/version.c
[tintin] LD build/tintin/tintin_fw.elf
132 bytes of RAM available for heap.
107544 bytes of flash unused.
OBJCOPY build/tintin/tintin_fw.bin
[tintin] PBZ build/tintin/tintin.pbz
[chalk] CC build/version.c
[chalk] LD build/chalk/tintin_fw.elf
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: build/chalk/tintin_fw.elf section `.ccmram' will not fit in region `RAM2'
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: region `RAM2' overflowed by 3412 bytes
collect2: error: ld returned 1 exit status
Makefile:156: recipe for target 'build/chalk/tintin_fw.elf' failed
make: *** [build/chalk/tintin_fw.elf] Error 1
```

I've used Linux Mint 19.1 based on Ubuntu.

GCC version:

```
(.env) a@a-VirtualBox:~/RebbleOS$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 7.3.0-27ubuntu1~18.04' --with-bugurl=file:///usr/share/doc/gcc-7/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-7 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 7.3.0 (Ubuntu 7.3.0-27ubuntu1~18.04) 
```

Full output:

```make
(.env) a@a-VirtualBox:~/RebbleOS$ make
[snowy] PBPACK_EXTRACT
512000+0 records in
512000+0 records out
512000 bytes (512 kB, 500 KiB) copied, 1,59287 s, 321 kB/s
[snowy] MKPACK res/snowy.json
wrote build/snowy/res/snowy_res.pbpack (190408 bytes)
wrote build/snowy/res/snowy_res.h
wrote build/snowy/res/snowy_res.d
VERSION build/version.c
[snowy] CC build/version.c
[snowy] CC FreeRTOS/croutine.c
[snowy] CC FreeRTOS/event_groups.c
[snowy] CC FreeRTOS/list.c
[snowy] CC FreeRTOS/queue.c
[snowy] CC FreeRTOS/tasks.c
[snowy] CC FreeRTOS/timers.c
[snowy] CC FreeRTOS/portable/GCC/ARM_CM4F/port.c
[snowy] CC FreeRTOS/portable/MemMang/heap_4.c
[snowy] CC lib/minilib/minilib.c
[snowy] CC lib/minilib/sbrk.c
[snowy] CC lib/minilib/dprint.c
[... 811 lines cut ...]
hw/platform/snowy_family/snowy_power.c:281:20: warning: 'return' with a value, in function returning void
             return 0;
                    ^
hw/platform/snowy_family/snowy_power.c:274:6: note: declared here
 void max14690_stay_on(uint8_t pullup)
      ^~~~~~~~~~~~~~~~
[chalk] CC hw/platform/snowy_family/snowy_scanlines.c
[chalk] CC hw/platform/snowy_family/snowy_vibrate.c
[chalk] CC hw/platform/snowy_family/snowy_ambient.c
[chalk] CC hw/platform/snowy_family/snowy_ext_flash.c
[chalk] CC hw/platform/snowy_family/snowy_common.c
[chalk] CC hw/platform/chalk/chalk.c
[chalk] CC hw/platform/chalk/chalk_bluetooth.c
hw/platform/chalk/chalk_bluetooth.c: In function 'hw_bluetooth_init':
hw/platform/chalk/chalk_bluetooth.c:10:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
[chalk] FPGA Resources/chalk_fpga.bin
[chalk] LD build/chalk/tintin_fw.elf
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: build/chalk/tintin_fw.elf section `.ccmram' will not fit in region `RAM2'
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: region `RAM2' overflowed by 3412 bytes
collect2: error: ld returned 1 exit status
Makefile:156: recipe for target 'build/chalk/tintin_fw.elf' failed
make: *** [build/chalk/tintin_fw.elf] Error 1
```

jwise, 2019-07-29:
Yep, that's a known issue right now building for `chalk`.  To work around it, just `make snowy` or `make tintin` for the time being.

Bystroushaak, 2019-07-29:
Oh, ok. So chalk is default and snowy / tintin has to be specified. Maybe it would be better to add this info to the guides? I'll send a pull request when I'll have some time.

Bystroushaak, 2019-07-29:
Oh, I just noticed that you added a lot of labels. Maybe I leave the issue open?

jwise, 2019-07-29:
Yeah, by default, we build all platforms (and, ideally, they all should build, but...).  If you specify just one, then you build only that one.

ginge, 2019-12-16:
Confirmed compile is clean and all is "working"

closed 2019-12-16 by ginge