pebble-dev/RebbleOS #94: Issue when built using cmd "make snowy"

kchinnasamy, 2020-02-23:
When I try to build the os, I was getting the below error, 
>  % make snowy
VERSION build/version.c
[snowy] CC build/version.c
[snowy] LD build/snowy/tintin_fw.elf
/mnt/c/Users/tuiszt/Documents/Pebble/ubuntu/pebble-dev/pebble-sdk-4.5-linux64/arm-cs-tools/bin/../lib/gcc/arm-none-eabi/4.7.2/../../../../arm-none-eabi/bin/ld: error: build/snowy/Resources/snowy_fpga.o: Conflicting CPU architectures 13/0
/mnt/c/Users/tuiszt/Documents/Pebble/ubuntu/pebble-dev/pebble-sdk-4.5-linux64/arm-cs-tools/bin/../lib/gcc/arm-none-eabi/4.7.2/../../../../arm-none-eabi/bin/ld: error: /mnt/c/Users/tuiszt/Documents/Pebble/ubuntu/pebble-dev/pebble-sdk-4.5-linux64/arm-cs-tools/bin/../lib/gcc/arm-none-eabi/4.7.2/thumb2/libgcc.a(_arm_muldivdf3.o): Unknown CPU architecture
/mnt/c/Users/tuiszt/Documents/Pebble/ubuntu/pebble-dev/pebble-sdk-4.5-linux64/arm-cs-tools/bin/../lib/gcc/arm-none-eabi/4.7.2/../../../../arm-none-eabi/bin/ld: error: /mnt/c/Users/tuiszt/Documents/Pebble/ubuntu/pebble-dev/pebble-sdk-4.5-linux64/arm-cs-tools/bin/../lib/gcc/arm-none-eabi/4.7.2/thumb2/libgcc.a(_aeabi_ldivmod.o): Unknown CPU architecture
/mnt/c/Users/tuiszt/Documents/Pebble/ubuntu/pebble-dev/pebble-sdk-4.5-linux64/arm-cs-tools/bin/../lib/gcc/arm-none-eabi/4.7.2/../../../../arm-none-eabi/bin/ld: error: /mnt/c/Users/tuiszt/Documents/Pebble/ubuntu/pebble-dev/pebble-sdk-4.5-linux64/arm-cs-tools/bin/../lib/gcc/arm-none-eabi/4.7.2/thumb2/libgcc.a(bpabi.o): Unknown CPU architecture
/mnt/c/Users/tuiszt/Documents/Pebble/ubuntu/pebble-dev/pebble-sdk-4.5-linux64/arm-cs-tools/bin/../lib/gcc/arm-none-eabi/4.7.2/../../../../arm-none-eabi/bin/ld: error: /mnt/c/Users/tuiszt/Documents/Pebble/ubuntu/pebble-dev/pebble-sdk-4.5-linux64/arm-cs-tools/bin/../lib/gcc/arm-none-eabi/4.7.2/thumb2/libgcc.a(_divdi3.o): Unknown CPU architecture
/mnt/c/Users/tuiszt/Documents/Pebble/ubuntu/pebble-dev/pebble-sdk-4.5-linux64/arm-cs-tools/bin/../lib/gcc/arm-none-eabi/4.7.2/../../../../arm-none-eabi/bin/ld: error: /mnt/c/Users/tuiszt/Documents/Pebble/ubuntu/pebble-dev/pebble-sdk-4.5-linux64/arm-cs-tools/bin/../lib/gcc/arm-none-eabi/4.7.2/thumb2/libgcc.a(_udivsi3.o): Unknown CPU architecture
4796 bytes of CCRAM available.
30056 bytes of RAM available for heap.
566604 bytes of flash unused.
OBJCOPY build/snowy/tintin_fw.bin
[snowy] PBZ build/snowy/snowy.pbz

System details 

> No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 19.10
Release:        19.10
Codename:       eoan

Please note I'm running window linux subsystem

it looks the arm-cs-tools is out date, not sure.

jwise, 2020-02-23:
Yeah, 4.7.2 is truly ancient.  Try using `gcc-arm-none-eabi` from apt, and then use the defaults rather than overriding `PEBBLE_TOOLCHAIN_PATH` in `localconfig.mk`?

ishotjr, 2020-02-23:
@kchinnasamy the `apt` version is likely ancient too - I haven't used *WSL* in a while, but here's my notes from a previous time I set it up:

---

Manually installed [ARM toolchain](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm):  

```bash
cd ~
wget https://developer.arm.com/-/media/Files/downloads/gnu-rm/8-2018q4/gcc-arm-none-eabi-8-2018-q4-major-linux.tar.bz2
mkdir -p "${HOME}/opt"
cd "${HOME}/opt"
tar xjf ~/gcc-arm-none-eabi-8-2018-q4-major-linux.tar.bz2
chmod -R -w "${HOME}/opt/gcc-arm-none-eabi-8-2018-q4-major"
"${HOME}/opt/gcc-arm-none-eabi-8-2018-q4-major/bin/arm-none-eabi-gcc" --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.1 20181213 (release) [gcc-8-branch revision 267074]
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
cat gcc-arm-none-eabi-8-2018-q4-major/share/doc/gcc-arm-none-eabi/readme.txt | more
rm ~/gcc-arm-none-eabi-8-2018-q4-major-linux.tar.bz2
```
*see also [GNU MCU Eclipse](https://gnu-mcu-eclipse.github.io/toolchain/arm/install/#gnulinux-1) :heart:*  

---

I tend to have to use a bunch of different versions for different projects so I will often end up just doing e.g. `export PATH="${HOME}/opt/gcc-arm-none-eabi-7-2017-q4-major/bin:$PATH"` as needed. Anyway, LMK if that doesn't fix it - I have a bunch of Windows machines that I've used for this in the past and a bunch of notes about how I got it working... :bowtie:

kchinnasamy, 2020-02-24:
Thank you both. 

I'm new to Firmware development, 

after installing the `gcc-arm-none-eabi` now I'm getting the below output 

` % make snowy
VERSION build/version.c
[snowy] CC build/version.c
[snowy] LD build/snowy/tintin_fw.elf
4796 bytes of CCRAM available.
30056 bytes of RAM available for heap.
568156 bytes of flash unused.
OBJCOPY build/snowy/tintin_fw.bin
[snowy] PBZ build/snowy/snowy.pbz`

Just wanna make sure if this works. 

and also I'm trying to run the firmware using `make snowy_qemu`, 

 % make snowy_qemu
make: *** No rule to make target 'Resources/snowy_boot.bin', needed by 'build/snowy/fw.qemu_flash.bin'.  Stop.

Am I missing something ?

AliceGrey, 2020-02-28:
> Thank you both.
> 
> I'm new to Firmware development,
> 
> after installing the `gcc-arm-none-eabi` now I'm getting the below output
> 
> ```
> % make snowy
> VERSION build/version.c
> [snowy] CC build/version.c
> [snowy] LD build/snowy/tintin_fw.elf
> 4796 bytes of CCRAM available.
> 30056 bytes of RAM available for heap.
> 568156 bytes of flash unused.
> OBJCOPY build/snowy/tintin_fw.bin
> [snowy] PBZ build/snowy/snowy.pbz
> ```
> 
> Just wanna make sure if this means the build is successful.
> 
> and also I'm trying to run the firmware using `make snowy_qemu`,
> 
> ```
> % make snowy_qemu
> make: *** No rule to make target 'Resources/snowy_boot.bin', needed by 'build/snowy/fw.qemu_flash.bin'.  Stop.
> ```
> 
> Am I missing something ?

It looks like you are missing the qemu resources.

> If you wish to run the firmware in qemu, copy the resources necessary into Resources/. Take a look at Utilities/mk_resources.sh for more information on that.

Also you will need the fpgas if you don't have them already

> You need the snowy_fpga.bin and chalk_fpga.bin files to compile on their respective firmwares. They can be found on the #firmware channel in the Rebble Discord.