pebble-dev/RebbleOS #99: Setting up pebble error

YonasBerhe, 2020-03-14:
[image]

UDXS, 2020-03-14:
This is not the correct place to report this issue. Please join the [Rebble Discord server here](http://discord.gg/aRUAYFN) and ask in #rebble-help.

Willow-Systems, 2020-03-14:
Hi @YonasBerhe. The issue you're seeing is because the app you're using is trying to load rebble over plaintext HTTP, which isn't allowed for auth. Please not that the 'rebble app' on the Google Play store is not an official rebble app, and therefore won't (and can't) be fixed by us.

For the official guide to setting up your android device, please see https://rebble.io/howto

closed 2020-03-14 by YonasBerhe

YonasBerhe, 2020-03-14:
Got it to work by not using the play store version of rebble for pebble and instead using the rebble docs