pebble-dev/RebbleOS #147: Revisit MenuLayer wraparound change (#145)

jwise, 2021-02-27:
@Katharine notes that #145 perhaps ought to be opt-in for apps.  We probably ought to revisit this at some point in the future...

Katharine, 2021-02-27:
To clarify for the mysterious future when someone looks at this issue, by "opt in" I mean "opt in by building against the Rebble SDK with a higher SDK version", not (necessarily) by making it a flag they can actively control.

If there _is_ an explicit API for it, it should probably default on for apps built with the appropriate SDK.

jwise, 2021-02-27:
Yeah, I think classically the way Pebble has handled this is by having the init routine change name.  We could do that too.  You know, in the Rebble SDK future.